ARMmbed/mbed-hal-st-stm32f4 #10: dependencie faild on st-nucleo-f401re

vahid-sohrabloo, 2015-09-08:
mbed-hal-st-stm32f401re module  not in yotta repository

~~~
error: Failed to satisfy dependencies of /home/vahid/blinky/yotta_modules/mbed-hal-st-stm32f4:
error: mbed-hal-st-stm32f401re does not exist in the modules registry. Check that the name is correct, and that it has been published.
~~~

0xc0170, 2015-09-08:
The target you are using is not fully supported, as the message indicates - modules are not published. Means the missing modules are not in the registry, neither on ARMmbed github organization. 
Please look at https://github.com/ARMmbed/mbed-os, where is the list of targets supported.

vahid-sohrabloo, 2015-09-08:
When this target is supported? and how I can help to support this target.

rainierwolfcastle, 2015-10-15:
http://jira-dev0.cambridge.arm.com/browse/IOTSFW-833

rainierwolfcastle, 2015-10-16:
ARM Internal Ref: IOTSFW-989

hugovincent, 2015-10-26:
Hi @vahid-sohrabloo, hopefully that target will be supported soon, but we don't currently have a timeframe for it.

0xc0170, 2015-11-24:
@vahid-sohrabloo This should be resolved. If this persists, reopen the issue or report a new one.

closed 2015-11-24 by 0xc0170