ARMmbed/mbed-hal-st-stm32f4 #12: `spi_frequency` has up to 50% error

salkinium, 2015-09-10:
There are several issues with the [`spi_frequency(spi_t *obj, int hz)`](https://github.com/ARMmbed/mbed-hal-st-stm32f4/blob/master/source/spi_api.c#L200-L305) method in `spi_api.c`:

The three points are exemplified with the following code, [lines 203 - 220](https://github.com/ARMmbed/mbed-hal-st-stm32f4/blob/master/source/spi_api.c#L203-L220):
```cpp
    // Note: The frequencies are obtained with SPI1 clock = 84 MHz (APB2 clock)
    if (hz < 600000) {
        obj->br_presc = SPI_BAUDRATEPRESCALER_256; // 330 kHz
    } else if ((hz >= 600000) && (hz < 1000000)) {
        obj->br_presc = SPI_BAUDRATEPRESCALER_128; // 656 kHz
    } else if ((hz >= 1000000) && (hz < 2000000)) {
        obj->br_presc = SPI_BAUDRATEPRESCALER_64; // 1.3 MHz
    } else if ((hz >= 2000000) && (hz < 5000000)) {
        obj->br_presc = SPI_BAUDRATEPRESCALER_32; // 2.6 MHz
    } else if ((hz >= 5000000) && (hz < 10000000)) {
        obj->br_presc = SPI_BAUDRATEPRESCALER_16; // 5.25 MHz
    } else if ((hz >= 10000000) && (hz < 21000000)) {
        obj->br_presc = SPI_BAUDRATEPRESCALER_8; // 10.5 MHz
    } else if ((hz >= 21000000) && (hz < 42000000)) {
        obj->br_presc = SPI_BAUDRATEPRESCALER_4; // 21 MHz
    } else { // >= 42000000
        obj->br_presc = SPI_BAUDRATEPRESCALER_2; // 42 MHz
    }
```

#### 1. Some decision frequencies are not consistent with the comments

It is clear that the decision frequencies do not match the ones from the comments and are therefore not correct.
This is, however, irrelevant because of point 3.

#### 2. The method is hardcoded to work for the max. CPU

It is clear that the code does not work properly for APB1/2 frequencies other than the maximum.
This can be fixed by computing the correct decision frequencies depending on the **actual** CPU frequency.

#### 3. The computed prescaler values are not the ones with the **least** amount of error!

This is actually the most problematic of the problems.
The code above, even if 1 and 2 are fixed, does not compute the prescaler values which corresponds to a frequency with the least amount of error.

Example: 

| Input | Output | Error (%) |
|---|---|---|
| 9MHz | 5.25MHz | -42% |
| 9.9MHz | 5.25MHz | -47% |
| 10.49MHz | 5.25MHz | -50% |
| 10.5MHz | 10.5MHz | 0% |

This is since the code does not consider the frequency above it, and therefore for 10.49MHz does not choose 10.5MHz which but 5.25MHz which is a massive 50% error!

#### Solution

The solution to this is actually very simple, since this is a power-of-two prescaler (i.e. 1,2,4,8,16), you can use some maths and arrive at the following solution:

You compute the desired divider:
```cpp
float prescaler = apb_frequency / hz;`
```
And then compare it to `4/3 * 2^n`:
```cpp
if (prescaler >= 256 * 4.f/3) {
    obj->br_presc = SPI_BAUDRATEPRESCALER_256;
} else if (prescaler >= 128 * 4.f/3) {
    obj->br_presc = SPI_BAUDRATEPRESCALER_128;
}
...
else {
    obj->br_presc = SPI_BAUDRATEPRESCALER_2;
}
```

You do not have to use floating point operations for this, you can also use a fixed point version, or even better use binary scaling to do this thing entirely in integer arithmetics.

This solution works for all power-of-two prescalers. I leave figuring out the math as an exercise for the reader. (The impatient can google the solution).

rainierwolfcastle, 2015-10-15:
http://jira-dev0.cambridge.arm.com/browse/IOTSFW-832

rainierwolfcastle, 2015-10-16:
ARM Internal Ref: IOTSFW-988