ARMmbed/mbed-hal-st-stm32f4 #20: LP timer is running ~1.25x too fast

niklas-arm, 2015-10-21:
The LP timer is running about 1.25 times too fast.

There is also a bug that a frequency is divided by a 1GHz value, which is of course nonsense:
https://github.com/ARMmbed/mbed-hal-st-stm32f4/blob/master/source/lp_ticker.c#L48

betzw, 2015-10-21:
A fix for this can be found in https://github.com/betzw/mbed-hal-st-stm32f4/tree/betzw_wb

0xc0170, 2015-10-21:
@betzw Are you planning to send a fix ?

betzw, 2015-10-21:
We are currently working together with @screamerbg on a port to Nucleo-STM32F401RE and have a first version pushed to github. I do niot know how @screamerbg (Mihail Stoyanov) is planning to proceed on this?

0xc0170, 2015-10-21:
lp ticker should be implemented on top of RTC, and this high-precision timer released for other use. 

@betzw  Are there any plans for this?

betzw, 2015-10-21:
In the mid-term, the answer is that using a RTC is clearly the right choice.
But currently we are all focused on TechCon demos ...

niklas-arm, 2015-10-21:
@betzw I don't understand your fix:
Doesn't Timer2 run from the APB1 clock domain, ie. which has a max. clock frequency limitation of 45MHz?
So the APB1 prescaler must be set to 4 (168MHz/4 = 42MHz < 45MHz), which means the Timer input clocks are multiplied by 2, ie. they run on 84MHz, aka. `SystemCoreClock / 2`?

Unfortunately the clock tree documentation in the F429 Reference Manual isn't very clear about which peripherals are powered by what clock domains.

niklas-arm, 2015-10-21:
@betzw please see the fix for this for you port of the F401.

simonqhughes, 2015-10-21:
ARM Internal Ref: IOTSFW-1065

betzw, 2015-10-22:
Well not sure I understand your comments. In any case let me explain my implementation:
The prescaler register of a timer has only 16-bits. The dafult SystemCoreClock on a NUCLEO-STM32F401RE is 84Mhz, which means that the value you need to put in the prescaler in order to obtain a 1ms tick is beyond the highest value you can put in the prescaler register (as it would require more than 16 bits). At that point I decide to put the highest possible prescaler value into the prescaler register in order to obtain the slowest tick ferquency possible and adjust the MINAR_PLATFORM_TIME_BASE value accordignly: https://github.com/betzw/mbed-hal-st-stm32f401re/blob/betzw_wb/mbed-hal-st-stm32f401re/target_config.h#L21.
Hope this helps.

PR #21 (2015-10-23): LP: Fix overflow in timer2 prescaler calculation.
This fixes the prescaler overflow and therefore the accuracy of the LP timer.

In all target configurations in the chip modules the `MINAR_PLATFORM_TIME_BASE` needs to be set to 2000 **NOT** 1000.

Fixes #20.
Depends on https://github.com/ARMmbed/mbed-hal-st-stm32f429zi/pull/15.

@bogdanm @bremoran @0xc0170
closed 2015-10-23 by 0xc0170