ARMmbed/mbed-hal-st-stm32f4 #27: Setting id=0 in gpio_irq_init() result in no interrupt callback

antevir, 2015-11-18:
If you init a irq gpio with id set to 0 like below, the handler will not be called:
```C
gpio_irq_init(&gpio, pin, handler, 0) 
````

The reason is the last line below in gpio_irq_api.c:
```C
static void handle_interrupt_in(uint32_t irq_index, uint32_t max_num_pin_line)
{
     ...
            // Clear interrupt flag
            if (__HAL_GPIO_EXTI_GET_FLAG(pin) != RESET) {
                __HAL_GPIO_EXTI_CLEAR_FLAG(pin);

                if (gpio_channel->channel_ids[gpio_idx] == 0) continue;
```

betzw, 2015-11-18:
But `gpio_irq_init()` is not a function of the mbed API which gets exported to the user but only used from inside the mbed implementation. I do **not** think that within mbed `gpio_irq_init()` gets ever called with an ''id' of zero.

0xc0170, 2015-11-18:
Can id be 0? The default state is all ids are 0. We should state this in the function documentation. I'll send PR.

antevir, 2015-11-18:
The reason we are using this API instead fo the higher level is for issue #28

antevir, 2015-11-19:
Okay, this was probably a missunderstanding from my side. But it would be nice if the documentation could be updated. 

An idea would also be to change the type of id to (void*) as the InterruptIn class uses it as a pointer. This will allow porting the HAL layer to platforms that are not 32bit-based (for instance linux target etc).

closed 2015-11-19 by 0xc0170