ARMmbed/mbed-hal-st-stm32f4 #32: Serial interrupt RX issue for stm32f4 devices (baud 115k, serial-interrupt test)

0xc0170, 2016-01-28:
mbed-drivers serial interrupt test is failing

```
HOST: Reset target...
HOST: Detecting test case properties...
HOST: Error! No host test name defined in preamble

{{error}}
{{end}}
mbedgt: mbed-host-test-runner: stopped
mbedgt: mbed-host-test-runner: returned 'ERROR'
mbedgt: test on hardware with target id: 072002210162647E3E26F263
        test 'mbed-drivers-test-serial_interrupt' .............................................. ERROR in 4.27 sec
mbedgt: shuffle seed: 0.1166601680
mbedgt: test report:
+----------------------+---------------+------------------------------------+--------+--------------------+-------------+
| target               | platform_name | test                               | result | elapsed_time (sec) | copy_method |
+----------------------+---------------+------------------------------------+--------+--------------------+-------------+
| st-nucleo-f401re-gcc | NUCLEO_F401RE | mbed-drivers-test-serial_interrupt | ERROR  | 4.27               | shell       |
+----------------------+---------------+------------------------------------+--------+--------------------+-------------+

+----------------------+------------------+------------------------------------+---------+--------------------+-------------+
| target               | platform_name    | test                               | result  | elapsed_time (sec) | copy_method |
+----------------------+------------------+------------------------------------+---------+--------------------+-------------+
| stm32f429i-disco-gcc | DISCOVERY-F429ZI | mbed-drivers-test-serial_interrupt | TIMEOUT | 55.69              | shell       |
+----------------------+------------------+------------------------------------+---------+--------------------+-------------+

// k64f just to be certain it is passing for any other target
+-----------------+---------------+------------------------------------+--------+--------------------+-------------+
| target          | platform_name | test                               | result | elapsed_time (sec) | copy_method |
+-----------------+---------------+------------------------------------+--------+--------------------+-------------+
| frdm-k64f-armcc | K64F          | mbed-drivers-test-serial_interrupt | OK     | 3.92               | shell       |
+-----------------+---------------+------------------------------------+--------+--------------------+-------------+
```

st-nucleo-f401re-gcc - the host test does not even read the preamble, returns error.  
stm32f429i-disco-gcc - test runs but timeouts, can't read probably RX line? When I connect to the serial port, I can see host tests prints, but does not echo anything.

@bcostm @betzw

bcostm, 2016-01-28:
Strange... I've just relaunched all mbed-drivers tests on F401 and F429 and this test is ok for me.

0xc0170, 2016-01-29:
Thanks for trying, I'll retest again with my 2 boards. I saw the output on disco f429 for instance, but could not make RX to work.

 @bcostm How did you test? Please run ``` mbedgt -n mbed-drivers-test-serial_interrupt ```. I don't think this test is automatically run, something to check !