ARMmbed/mbed-hal-st-stm32f4 #35: Missing initialization of ADC clocks

andreaslarssonublox, 2016-03-02:
In function analogin_api.c:analogin_init only one ADC(ADC1) clock is initialized. Shouldn't all three be initialized(ADC1, ADC2,ADC3) if they are present e.g. in stm32f439?

        // Enable ADC clock
        __ADC1_CLK_ENABLE();

0xc0170, 2016-03-03:
@bcostm

bcostm, 2016-03-03:
Yes all the clocks of the used ADC you must be enabled. In the analogin_api.c file that I see on GitHub there are the clocks for ADC1 and ADC3. There is not the ADC2 clock because it is either not used or removed by someone or simply missing since the beginning. If you need it in your platform don't hesitate to add it by doing the same thing as for the other existing clocks. Thanks.

andreaslarssonublox, 2016-03-10:
Ok, yes I can do a pull-request with added ADC2 and ADC3. My concern though is that there might be other stm32f4 devices not supporting 3 ADCs but I guess the following can be used(derived from stm32f4xx_hal_rcc_ex.h):

        // Enable ADC clock
        __ADC1_CLK_ENABLE();
#if defined(STM32F427xx) || defined(STM32F437xx) || defined(STM32F429xx)|| defined(STM32F439xx) || defined(STM32F469xx) || defined(STM32F479xx)
        __ADC2_CLK_ENABLE();
        __ADC3_CLK_ENABLE();       
#endif

bcostm, 2016-03-11:
Please use ` #if defined(ADC2)` instead of using device names.

andreaslarssonublox, 2016-03-11:
Fixed in:
https://github.com/ARMmbed/mbed-hal-st-stm32f4/pull/40